nebrius/raspi #2: ERROR: could not insert 'i2c_dev': Operation not permitted

LegendaryTom, 2015-06-03:
Hello,

I'm trying ```raspi-io``` for the first time and am trying to run the "blink a led" script from the readme.  

Here is the script that I'm using:
```javascript
var raspi = require('raspi-io');
var five = require('johnny-five');
var board = new five.Board({
  io: new raspi()
});

board.on('ready', function() {

  // Create an Led on pin 7 (GPIO4) on P1 and strobe it on/off
  // Optionally set the speed; defaults to 100ms
  (new five.Led('GPIO18')).strobe();

});
```

When I run the script, I'm seeing the following error:
```
node j5_test_pi.js 
ERROR: could not insert 'i2c_dev': Operation not permitted
child_process.js:1382
    throw err;
          ^
Error: Command failed: modprobe i2c-dev
ERROR: could not insert 'i2c_dev': Operation not permitted

    at checkExecSyncError (child_process.js:1339:13)
    at execSync (child_process.js:1379:13)
    at new I2C (/home/pi/evcc/node_modules/raspi-io/node_modules/raspi-i2c/lib/index.js:115:5)
    at /home/pi/evcc/node_modules/raspi-io/lib/index.js:150:16
    at new Raspi (/home/pi/evcc/node_modules/raspi-io/lib/index.js:185:7)
    at Object.<anonymous> (/home/pi/evcc/j5_test_pi.js:4:7)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
```

It looks like the error happens on the line with ```new raspi()```.  

I then noticed that the output from ```npm install raspi-io``` had these lines:
```
> execSync@1.0.2 install /home/pi/evcc/node_modules/raspi-io/node_modules/raspi-i2c/node_modules/execSync
> node install.js

[execsync v1.0.2] Attempting to compile native extensions.
[execSync v1.0.2]
    Native code compile failed!!
```

Have I hosed my installation of ```raspi-io```?

My setup is a Raspberry Pi 2 Model B v1.1, with node v0.12.0, johnny-five@0.8.76, and raspi-io@3.2.2 installed.  I also tried raspi-io@3.1.1 and got a similar result.

bryan-m-hughes, 2015-06-03:
Are you running the script as root? If not, this is why you're getting the `Operation not permitted error`. Can you try that and let me know if everything works?

The execSync thing isn't an issue in your case. It's only there as a fallback for node 0.10, which doesn't support it natively.

LegendaryTom, 2015-06-03:
That was what I was doing wrong.  Thanks for your help!

closed 2015-06-03 by LegendaryTom

bryan-m-hughes, 2015-06-03:
Great to hear!