nebrius/raspi #4: What would be the best way to initialize raspi

gswcm, 2016-10-09:
I am trying to build a module based on raspi (to make a simple beep) that will be embedded into a bigger project. Here is the module's code:
```
var raspi = require('raspi');
var gpio = require('raspi-gpio');

var Scanner = function() {
	var self = this;
	raspi.init(function() {
		self.buzzer = new gpio.DigitalOutput('P1-32');
	});
};

Scanner.prototype.beep = function() {
	var self = this;
	self.buzzer.write(gpio.HIGH);
	setTimeout(function() {
		self.buzzer.write(gpio.LOW);
	}, 20);
};

module.exports = exports = Scanner;
```
and I am using it in a bigger project as 
```
var Scanner = require('./scanner');
var scanner = new Scanner();

scanner.beep();
```
but it throws me an error stating that 'buzzer' is not defined. I guess it is because constructor returns before **raspi** initialization is complete, so the beep() function is invalid. I tried to set a timeout before using beep():
```
var Scanner = new require('./scanner');
var scanner = new Scanner();

setTimeout(function(){
	scanner.beep();
}, 1000);
``` 
and it works... but I was wondering if there exist a better way to do it without timeout. 

I read a few discussions about asynchronous constructors that propose constructor to accept a callback to be called when all initialization is complete. Would this be the right way? 

Thank you.

nebrius, 2016-10-10:
Using a callback in a constructor would work, but that's not a common practice. I would recommend having an empty constructor coupled with an `init` method, like this:

```
var raspi = require('raspi');
var gpio = require('raspi-gpio');

var Scanner = function() {};

Scanner.prototype.init = function(cb) {
    var self = this;
    raspi.init(function() {
        self.buzzer = new gpio.DigitalOutput('P1-32');
        cb();
    });
}

Scanner.prototype.beep = function() {
    var self = this;
    self.buzzer.write(gpio.HIGH);
    setTimeout(function() {
        self.buzzer.write(gpio.LOW);
    }, 20);
};

module.exports = exports = Scanner;
```

and then the code using your module would look like:

```
var Scanner = new require('./scanner');
var scanner = new Scanner();

scanner.init(function() {
  scanner.beep();
});
```

gswcm, 2016-10-10:
Thank you, it is very much alike to what I ended up with after posting the
issue. Please close the case.

closed 2016-10-10 by nebrius