nebrius/raspi #5: Use wiringPiSetupSys to avoid running as root

RandScullard, 2016-10-24:
I would love to be able to access the GPIO pins without root access, and apparently this is possible (at least in theory). According to this article: https://www.raspberrypi.org/forums/viewtopic.php?f=34&t=51265 the key is calling wiringPiSetupSys instead of wiringPiSetup. Would it be possible to have the raspi module work this way?

nebrius, 2016-10-25:
We could, but I'm against it for two reasons:

1. `wiringPiSetypSys` requires that the gpio interfaces be exported prior to the application starts...which is an operation that requires root privileges. Requiring that users do an operation by hand before using raspi-io is going to create a lot of confusion. My goal with raspi-io is for it to be as plug and play as possible, and `wiringPiSetypSys` goes against that.

2. This changes the pin numbering scheme and would require some architectural changes in raspi-io and raspi-board, and would also be a breaking change for everyone that uses raspi-io.

Does that make sense?

RandScullard, 2016-10-25:
I was thinking of it being an option, for exactly the reasons you stated. I agree that you definitely wouldn't want to change the current behavior for existing users of the module.

nebrius, 2016-10-25:
> I was thinking of it being an option, for exactly the reasons you stated. I agree that you definitely wouldn't want to change the current behavior for existing users of the module.

🤔

Even as an option, there's still point number 2. Personally, I don't really want to implement it because I don't have much time I can dedicate to raspi-io these days, and this feature is a lot lower on my personal priority list than other things I want to implement (e.g. SPI, unit tests).

That said, this is a feature I wouldn't mind having if it were an option, and if someone (cough cough) were to submit a the necessary pull requests, I'd happily review it! Is this something you'd be interested in working on?

RandScullard, 2016-10-25:
I might be... I'm in the middle of a bunch of things right now, and by the time I'm done I may not care that I have to run these things as root. 😃

nebrius, 2017-06-02:
Raspi IO recently switched to using pigpio instead of Wiring Pi, so I'm going to go ahead and close this.

closed 2017-06-02 by nebrius

RandScullard, 2017-06-10:
@nebrius Can you explain briefly what drove your decision to move from Wiring Pi to pigpio? I have a project that currently uses Wiring Pi and I'm considering making the same move, so I'm very curious to know what your reasons were.

nebrius, 2017-06-12:
@RandScullard I don't know if this is the reason you're looking for, but check out https://github.com/nebrius/raspi-io/issues/91.

tl;dr Wiring Pi in combination with some still unknown other factors flat out broke Raspi IO, and I was never able to figure out why.

A nice side-effect though is that switching to pigpio allowed me to get rid of my own native build step, making Raspi IO 100% JavaScript.

RandScullard, 2017-06-12:
@nebrius Thanks for the info, I appreciate it!

I wonder if your issue https://github.com/nebrius/raspi-io/issues/91 was caused by this: https://github.com/WiringPi/WiringPi-Node/issues/70

I think I will go ahead and switch to pigpio as well, because of this issue as well as the apparent lack of active maintenance on WiringPi-Node.

nebrius, 2017-06-12:
Oh I remember that project, I briefly looked into it but decided maintaining my own native module was the better approach, although I can't remember exactly why off the top of my head.

RandScullard, 2017-06-26:
Just in case anyone else can benefit from this info...

I looked into using pigpio for my own project, but then I realized that it requires root access, which I want to avoid (see above). This (finally) forced me to RTFM and understand how the /sys/class/gpio interface works, and I realized that I can just use file system APIs to read and write GPIO pins (no root access required). I threw together a simple wrapper module and dumped WiringPi-Node... which is what I should have done in the first place!

(Note that my project doesn't require anything fancy like PWM, just low-frequency reading and writing of GPIO pins.)

nebrius, 2017-06-26:
Thanks for the info, and I'm glad you figured out a workaround @RandScullard!