nebrius/raspi #6: "compilation terminated" on Raspbian Raspberry Pi Zero

akinnee, 2017-02-12:
`node -v`: `v6.9.4`

```
pi@K-pi:~/repos/K-pi/johnny-five-lifx $ npm install raspi@3.0.5

> raspi@3.0.5 install /home/pi/repos/K-pi/johnny-five-lifx/node_modules/raspi
> node-gyp rebuild

gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
make: Entering directory '/home/pi/repos/K-pi/johnny-five-lifx/node_modules/raspi/build'
  CXX(target) Release/obj.target/addon/src-native/addon.o
  CXX(target) Release/obj.target/addon/src-native/init.o
../src-native/init.cc:25:22: fatal error: wiringPi.h: No such file or directory
 #include <wiringPi.h>
                      ^
compilation terminated.
addon.target.mk:93: recipe for target 'Release/obj.target/addon/src-native/init.o' failed
make: *** [Release/obj.target/addon/src-native/init.o] Error 1
make: Leaving directory '/home/pi/repos/K-pi/johnny-five-lifx/node_modules/raspi/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/pi/.nvm/versions/node/v6.9.4/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.34+
gyp ERR! command "/home/pi/.nvm/versions/node/v6.9.4/bin/node" "/home/pi/.nvm/versions/node/v6.9.4/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/repos/K-pi/johnny-five-lifx/node_modules/raspi
gyp ERR! node -v v6.9.4
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
npm WARN johnny-five-lifx@1.0.0 No description
npm WARN johnny-five-lifx@1.0.0 No repository field.
npm ERR! Linux 4.4.34+
npm ERR! argv "/home/pi/.nvm/versions/node/v6.9.4/bin/node" "/home/pi/.nvm/versions/node/v6.9.4/bin/npm" "install" "raspi@3.0.5"
npm ERR! node v6.9.4
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! raspi@3.0.5 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the raspi@3.0.5 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the raspi package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs raspi
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls raspi
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/repos/K-pi/johnny-five-lifx/npm-debug.log
```

nebrius, 2017-02-13:
Looks like you don't have wiring pi installed? Are you by chance running the lite version of raspbian?

You can fix this with `sudo apt-get install wiringpi`

akinnee, 2017-02-15:
That was it! Thanks. I am running the full version I believe. Here's the os release info:
`cat /etc/os-release`
```
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

closed 2017-02-15 by akinnee