nebrius/raspi #7: Error: Cannot find module 'raspi'

javirs85, 2017-03-06:
Im using raspberry 2 B+ with last version of raspbian. 
node -v => v7.6.0
npm -v =>4.1.2

Problem was that from the repo I cannot install node higher than v0.10 so I just dowloaded the bin from the file, overwrite and fixed the soft link in the bin folder. So far it worked like a charm : )

this is what I get when I install the package:

> pi@pegsPi:~/web $ sudo npm install -g raspi
> 
> > raspi@3.0.5 install /opt/nodejs/lib/node_modules/raspi
> > node-gyp rebuild
> 
> gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/7.6.0"
> gyp WARN EACCES attempting to reinstall using temporary dev dir "/opt/nodejs/lib/node_modules/raspi/.node-gyp"
> make: Entering directory '/opt/nodejs/lib/node_modules/raspi/build'
>   CXX(target) Release/obj.target/addon/src-native/addon.o
>   CXX(target) Release/obj.target/addon/src-native/init.o
>   SOLINK_MODULE(target) Release/obj.target/addon.node
>   COPY Release/addon.node
> make: Leaving directory '/opt/nodejs/lib/node_modules/raspi/build'
> /opt/nodejs/lib
> └── raspi@3.0.5
> 

this is the relevant part of my main.js

```
var express = require('express');
var app = express();

var server = require('http').Server(app);
var io = require('socket.io')(server);

const raspi = require('raspi');
const gpio = require('raspi-gpio');

```

and this is the outcome when I start

> > nodemon server/main.js
> 
> [nodemon] 1.11.0
> [nodemon] to restart at any time, enter `rs`
> [nodemon] watching: *.*
> [nodemon] starting `node server/main.js`
> module.js:472
>     throw err;
>     ^
> 
> Error: Cannot find module 'raspi'
>     at Function.Module._resolveFilename (module.js:470:15)
>     at Function.Module._load (module.js:418:25)
>     at Module.require (module.js:498:17)
>     at require (internal/module.js:20:19)
>     at Object.<anonymous> (/home/pi/web/server/main.js:7:15)
>     at Module._compile (module.js:571:32)
>     at Object.Module._extensions..js (module.js:580:10)
>     at Module.load (module.js:488:32)
>     at tryModuleLoad (module.js:447:12)
>     at Function.Module._load (module.js:439:3)
> [nodemon] app crashed - waiting for file changes before starting...

I'm new to node so it could be any stupid thing...

nebrius, 2017-03-06:
The issue is with your installation command. `sudo npm install -g raspi` should be `npm install raspi`. Note that we drop both the `sudo` and the `-g` bits.

When you install an npm module with the `-g` flag, it adds it to the global `node_modules` folder. However, node doesn't search the global `node_modules` folder when looking for modules, hence the error.

javirs85, 2017-03-07:
oh ... I see ..

I removed raspi and raspi-gpio from -g and installed both as you suggested (no -g no sudo )

This is what I get now: 

> wiringPiISR: wiringPi has not been initialised. Unable to continue.

I check that wiringpi is installed:

> pi@pegsPi:~/web $ sudo apt install wiringpi
> Leyendo lista de paquetes... Hecho
> Creando árbol de dependencias
> Leyendo la información de estado... Hecho
> wiringpi ya está en su versión más reciente.

How should I initialise wiringpi ? This is the relevant part of the code

```
const raspi = require('raspi');
const gpio = require('raspi-gpio');

server.listen(8080, function() {
const output = new gpio.DigitalOutput('GPIO18');
```

here it crash :(

nebrius, 2017-03-07:
Looks like you're missing the `raspi` initialization. Try:

```
const raspi = require('raspi');
const gpio = require('raspi-gpio');

raspi.init(() => {
  server.listen(8080, () => {
    const output = new gpio.DigitalOutput('GPIO18');
    // Do stuff with output
  });
});
```

javirs85, 2017-03-07:
Perfect! 

Now it does not complain anymore but .. my leds does not turn on :(

I´m using the raspi1 reb 2 and im using this file as reference: [https://github.com/nebrius/raspi-io/wiki/Pin-Information](https://github.com/nebrius/raspi-io/wiki/Pin-Information)

I connected a 150 Ohm resistor to GPIO24/5/P1-18. Connected an LED in series and to Ground (just the pin below it). The led is lighting when connected between 3V3 and GND. and the console.log shows that the code actually goes through.. Im pretty sure I´m messing it up when setting the name of the pin but .. it really looks okay :( 

Here is the code: 

```
var isLEDOn = false;
var output;

server.listen(8080, function() {
         raspi.init(() => {
	       output = new gpio.DigitalOutput('5');
	});
});

io.on('connection', function(socket) {  
     socket.on('turnLED1', function(data){
                if(isLEDOn){
			output.write(gpio.LOW);
                         console.log("LOW");
		}else{
			output.write(gpio.HIGH);
			console.log("HIGH");
		}
		isLEDOn = !isLEDOn;
	});
});
```

nebrius, 2017-03-07:
Ah, yes, if you want to use the wiring pi pin number, it needs to be a number not a string, e.g. `output = new gpio.DigitalOutput(5);` without the quotes around `5`. Alternately, you can do `output = new gpio.DigitalOutput('P1-18');` or `output = new gpio.DigitalOutput('GPIO24');` if it helps you to remember.

javirs85, 2017-03-08:
First and foremost I want to THANK you for the support I'm receiving. I never expected so much attention and I'm SO glad..  Thanks :)

Second, the error is slightly getting far away from the main topic, I'm not sure if you want me to open a new discussion since the loading of the raspi is already solved.

I will deeply test that tonight (Europe time), Thanks once again : )

javirs85, 2017-03-08:
Ok, I did a python version of the code, looks like this:

```
import RPi.GPIO as GPIO            # import RPi.GPIO module  
from time import sleep             # lets us have a delay  
GPIO.setmode(GPIO.BOARD)             # choose BCM or BOARD  
port = 23;
GPIO.setup(port, GPIO.OUT)           # set GPIO24 as an output   
  
try:  
    while True:  
        GPIO.output(port, 1)         # set GPIO24 to 1/GPIO.HIGH/True  
        sleep(0.5)                 # wait half a second  
        GPIO.output(port, 0)         # set GPIO24 to 0/GPIO.LOW/False  
        sleep(0.5)                 # wait half a second  
  
except KeyboardInterrupt:          # trap a CTRL+C keyboard interrupt  
    GPIO.cleanup()                 # resets all GPIO ports used by this program 
```

the LED blinks ... 

This what I have in the js: 

```
var express = require('express');
var app = express();

var server = require('http').Server(app);
var io = require('socket.io')(server);

const raspi = require('raspi');
const gpio = require('raspi-gpio');

var Light = require('./light.js');

app.use(express.static('public'));  

var isLEDOn = false;
var output;

server.listen(8080, function() {
	console.log('Servidor corriendo en http://localhost:8080');
	
	 //Sets port for reading
	raspi.init(() => {
		/*const input = new gpio.DigitalInput({
			pin: '12',
			pullResistor: gpio.PULL_UP
		});*/
		output = new gpio.DigitalOutput("P1-23");
		output.write(gpio.HIGH);
		console.log("P1-23 set to high");
	});
});

io.on('connection', function(socket) {  
    console.log('Un cliente se ha conectado');
    
	//sends message to clients
	//socket.emit('messages', messages);
	
	//receive message from client
	socket.on('turnLED1', function(data){
		if(isLEDOn){
			output.write(gpio.LOW);
			console.log("LOW");
		}else{
			output.write(gpio.HIGH);
			console.log("HIGH");
		}
		isLEDOn = !isLEDOn;
	});	
	
	socket.on('readPin', function(data){});
});
```

 nothing happens :(

any clue??

javirs85, 2017-03-08:
Im using a raspi 1, the one with short gpio (26 pins) can it be the reason ?

nebrius, 2017-03-08:
Can you do me a favor and run this code:

```
const raspi = require('raspi');
const gpio = require('raspi-gpio');

var output;

//Sets port for reading
raspi.init(() => {
  output = new gpio.DigitalOutput("P1-23");
  console.log(output.pins);
  console.log(gpio.HIGH);
  output.write(gpio.HIGH);
  console.log("P1-23 set to high");
});
```

The idea is to start with a simple case, get it working, and then start reintroducing things back.

> Im using a raspi 1, the one with short gpio (26 pins) can it be the reason ?

To clarify, are you using a Raspberry Pi Model B Rev 1?

javirs85, 2017-03-09:
I'm running Model **B Rev 2	512MB** (checked with cat /proc/cpuinfo)

I ran the code you proposed, this is the outcome:

```
pi@pegsPi:~/web/server $ sudo node test.js
[ 14 ]
1

```

I removed all electronics just to simplify the thing, I used my multimeter to measure between pin 23 and 25(GND) there is a weird 1v , nor zero nor 3.3. I checked against pin1 and there is a full 3.3v measurement there.
Ran again python script and in the multimeter I see it bouncing from 0 to 3.3

anything else I can provide to you ?

nebrius, 2017-06-02:
Ack, I kinda let this slip through the cracks. Were you able to get it working, and can we close this issue now?

closed 2017-06-02 by nebrius