nebrius/raspi #9: Can't install on Windows

chschafl, 2018-05-26:
Unfortunately the yarn add fails for me on Windows:

`error raspi@5.0.2: The CPU architecture "x64" is incompatible with this module.
error An unexpected error occurred: "Found incompatible module".`

Would it be possible to still allow install on Windows and run in "mock" mode?
This would it make easier to develop for RPI.

Or am I doing something wrong?

nebrius, 2018-05-29:
The behavior your seeing is expected, as Raspi IO is designed to run on Linux distributions running on a Raspberry Pi only.

Raspi IO doesn't have any sort of mock mode and implementing it would take more work than I have time for due to internal coupling/architecture. 

You could try using [Board IO](https://github.com/achingbrain/board-io) to create your own mock class though and only include Raspi IO when your on a Raspberry Pi.

nebrius, 2018-06-21:
I'm going to go ahead and close this issue, but feel free to reopen if you have further questions.

closed 2018-06-21 by nebrius