nebrius/raspi #10: Installation directory depreciated

anders-wal, 2018-07-27:
When trying to install under the latest Raspbian, the installation aborts with the following output. It seems that the installation script needs tweaking.

> npm install raspi
> (node:15138) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
> npm WARN install Refusing to install raspi as a dependency of itself
>

anders-wal, 2018-07-27:
Addendum:

After a reinstallation of Node and NPM, the install of raspi still fails, but in a different way:

> $ sudo npm install raspi
> npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
> npm notice created a lockfile as package-lock.json. You should commit this file.
> npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
> npm WARN pi No description
> npm WARN pi No repository field.
> npm WARN pi No README data
> npm WARN pi No license field.
> 
> + raspi@5.0.2
> updated 1 package in 1.282s
>

nebrius, 2018-07-27:
The first issue sounds like some weird fluke with your node install, the second is actually fine. Those warnings can be ignored, and the last bit (raspi@5.0.2 updated 1 package in 1.282s) indicates that it installed OK.

Can you try running your app and seeing if it works?

nebrius, 2018-08-20:
I'm going to go ahead and close this issue due to inactivity. If you still need help, feel free to reopen it.

closed 2018-08-20 by nebrius