nebrius/raspi #11: raspi doesn't work consistently

malhar-quickwork, 2018-08-17:
I'm using raspi to control a motor but the module doesn't always connect to the motor.
I've made a test script which runs perfectly every time, but when I implement the code on server it doesn't always connect 
Test script that works everytime
```
const raspi = require('raspi');
const PWM = require('raspi-pwm').PWM;
let speed = 9 ;
if(process.argv[2] ){
    speed = Number(process.argv[2])
    console.log('Speed ',speed);
    raspi.init(() => {
      let pin2 = 'GPIO13';
      const pwm2 = new PWM(pin2);
      pwm2.write(speed); // Center a servo
      console.log('speed On GPIO13 : ',speed);
    });
}else{
  console.log('fail')
}
```

server code
```
const raspi = require('raspi');
const PWM = require('raspi-pwm').PWM;
let pwm12,pwm13,pwm18;
let self = {
    init: () =>{
        raspi.init(() => {
            pwm12 = new PWM('GPIO12');
            pwm13 = new PWM('GPIO13');
            pwm18 = new PWM('GPIO18');
            self.motorsArr.push(pwm13);            
            self.motorsArr.push(pwm12);
            self.motorsArr.push(pwm18);
            console.log(self.motorsArr[1]);
        });}
    throttle: (dutyCycle) {
        pwm13.write(dutyCycle);
}
}
```

Sometimes though it works after closing and reopening the terminal

nebrius, 2018-08-17:
Ah, the issue is that you're trying to drive 3 PWM motors using 2 PWM ports. It's kinda confusing, but `GPIO12` and `GPIO18` are both tied to `PWM0`, and `GPIO13` and `GPIO19` are both tied to `PWM1`. If you want to use three pins as PWM, you'll need to set `enableSoftPwm` to `true` in the Raspi IO constructor.

malhar-quickwork, 2018-08-18:
I'm new to this, so can you please show a small snippet or point me to some examples

nebrius, 2018-08-20:
Ack, sorry, I just realized you're not using [Raspi IO](https://github.com/nebrius/raspi-io/), which is how most folks use this library instead of using it directly. OK, so ignore what I said above, sorry about the confusion 😅

Instead, what you'll need to do is use the [raspi-soft-pwm](https://github.com/nebrius/raspi-soft-pwm) so that you can use more than two PWM pins at a time. Try this:

```JavaScript
const raspi = require('raspi');
const PWM = require('raspi-soft-pwm').SoftPWM;
let pwm12,pwm13,pwm18;
let self = {
    init: () => {
        raspi.init(() => {
            pwm12 = new PWM('GPIO12');
            pwm13 = new PWM('GPIO13');
            pwm18 = new PWM('GPIO18');
            self.motorsArr.push(pwm13);            
            self.motorsArr.push(pwm12);
            self.motorsArr.push(pwm18);
            console.log(self.motorsArr[1]);
        });
    },
    throttle: (dutyCycle) => {
        pwm13.write(dutyCycle);
    }
}
```

Note: raspi-soft-pwm _emulates_ PWM using some clever memory tricks on the Raspberry Pi. It works just fine for servo motors, but you will see slightly increased CPU usage.

malhar-quickwork, 2018-08-21:
Thank you. But can you add a terminate() like function like the pigpio library has. I think problem was caused by the pigpio C library not terminating when my script terminates

closed 2018-08-21 by malhar-quickwork

nebrius, 2018-09-07:
> Thank you. But can you add a terminate() like function like the pigpio library has. I think problem was caused by the pigpio C library not terminating when my script terminates

Sorry for the delay, I've been at Burning Man and offline for a long time. Can you clarify what behavior you're seeing that makes you think the above is happening?

The way we're using pigpio, it terminates as soon as the Node process does, because we're not using the daemon and it runs in the same process. There's nothing that keeps on running, but perhaps something is being left in a state you wouldn't expect?